stuartpittaway/diyBMS #3: PCB works - almost!

stuartpittaway, 2017-11-26:
I've put together version 1 of the PCB and it appears to work okay - but not as reliable as my breadboard version.

The i2c traffic works for several requests then appears to skip a few requests and then works again for a few more etc.  The breadboard version is rock solid.

The ADUM1250 chip took a bit of a bashing when soldering, so this may be the weak point - although it does work intermittently.

chickey, 2017-11-26:
Hopefully won't take too long for my boards and surface mount attiny's to
arrive.  Thinking about it i could breadboard one up quite easily minus the
ADUM with it being surface mount.

stuartpittaway, 2017-11-26:
I used some SOIC adapter boards which work quite well.

stuartpittaway, 2017-11-26:
Panic over, nothing wrong with the board or ADUM chip - I was using an MCP1702 regulator instead of the 1703A but only feeding the board with 3.3V from supply which caused it to brown out every now and again.

I also think I need to remove the RED led circuit components as the board is pulling a few extra mA current with that resistor network always connected.

stuartpittaway, 2017-11-27:
Second panic over, looks like I killed the ATTINY chip during soldering or testing - the ADC on Pin 2 appeared to work, but the reading would slowly drop in value over a few minutes.

Removed and replaced the ATTINY and works fine now.

closed 2017-11-27 by stuartpittaway