stuartpittaway/diyBMS #6: Is resistor R2 too low?

stuartpittaway, 2018-04-30:
Original design had R2 as 47 ohm

Using..
https://www.digikey.co.uk/en/resources/conversion-calculators/conversion-calculator-led-series-resistor

4.2V supply, 3.3V forward voltage (blue LED) and 20mA current gives 45ohm resistor - so the 47 is a logical choice.

Note that R2 also adds to the balance load on the cells in addition to the load resistor.

What do others think?

CuriousTimo, 2018-04-30:
> calculator-led-series-resistor

Hi Stuart,

Adam Welch explains it well in his YouTube video
https://youtu.be/ETw4SNH7SXI?t=605
His video was what decided to switch to a 510R resistor.

Cheers,

Timo

chickey, 2018-04-30:
With 510ohm, wouldn't that give just over 2ma which seems really low.

AdamWelchUK, 2018-05-01:
Hello all.  I guess this is mentioned because of my mistake.  I forgot to factor in the voltage drop of the LED into my calculation and suggested R2 was too low in my video.  So firstly Stuart, sorry about that, I hope you will forgive me.

As you know I've put a 510ohm in for R2, which as you say Colin gives about 2mA.  Now, although that original decision is based on incorrect maths I'm going to stick with 510 ohm.  I've attached two images which I took with fixed settings on my camera a few seconds apart.  One is running at 2mA the other 20mA.  It's hard to reproduce light levels on camera, but it should give you a rough idea.

[image]
[image]

To add to the discussion I'll argue that the value of R2 will entirely depend on each users individual situation.  A system placed in a dim environment rather than a sunny one is one factor, but the amount of time the LED is likely to be lit is another. My LEDs were very cheap and I'd rather not have to replace them, running them at a low current should help that considerably.  I've also swapped the suggested blue LED for red - with a lower forward voltage.  Red makes more sense to me as an indicator to suggest something is wrong (high) or we're burning off the energy.
Now I'll also argue that R2 may be chosen with consideration of R1.  If you are aiming for a discharge current of 200mA for example (say with a 20ohm R1) then an additional 20mA is an additional 10%, but if you are going for a ~1amp discharge current (eg. R1 is 4.3 ohm), then 20mA extra is almost nothing. 

Cheers

stuartpittaway, 2018-06-11:
Thanks for the post @AdamWelchUK closing this issue now as you explain the reasons why very well :-)

closed 2018-06-11 by stuartpittaway