stuartpittaway/diyBMS #9: New PCB

tvixen, 2018-07-12:
Hi Stuart

I made a new PCB for you. You can have a look and if its ok, approve it.  
This one is made with THT components, as many people don't have SMD in the drawer

[DIY BMSP Ver 3.0 brd .pdf](https://github.com/stuartpittaway/diyBMS/files/2188237/DIY.BMSP.Ver.3.0.brd.pdf)
[DIY BMSP Ver 3.0 sch .pdf](https://github.com/stuartpittaway/diyBMS/files/2188238/DIY.BMSP.Ver.3.0.sch.pdf)

Br Tim

tvixen, 2018-07-12:
And here goes the files: 
[DIY BMSP Ver 3.0 brd .pdf](https://github.com/stuartpittaway/diyBMS/files/2188281/DIY.BMSP.Ver.3.0.brd.pdf)
[DIY BMSP Ver 3.0 sch .pdf](https://github.com/stuartpittaway/diyBMS/files/2188282/DIY.BMSP.Ver.3.0.sch.pdf)

brunkj, 2018-07-12:
nice!  What is the physical dimensions of the board?

tvixen, 2018-07-12:
78mm x 48mm

stuartpittaway, 2018-07-12:
Nice work!  Do you need the capacitors on the inputs/outputs of the ADUM1250 module ?

tvixen, 2018-07-12:
I would say, it depends of the length on side 2. But you will have a nicer risetime, and that might eliminate some i2c errors.

closed 2019-03-04 by stuartpittaway