stuartpittaway/diyBMS #10: ESP code conver to Arduino Uno code

Rorschac, 2018-07-16:
Hi, I want to convert "ESP test" code  to arduino uno code. I have removed some code lines specified for ESP8266, and I compiled with any problem for Arduino uno.  Now, I have tested this code that i send "X" from serial interface. I see the message on I2C, but attiny85 did not response.  I dont now where I did wrong. I added scope screen and that i removed code lines.

```
extern "C"
{
  #include "user_interface.h"
}

os_timer_t myTimer;

//  os_timer_setfn(&myTimer, timerCallback, NULL);
//  os_timer_arm(&myTimer, 1000, true);
```
[image]

tvixen, 2018-07-16:
Hi there
If it helps, you can have a look at my code. 
I converted it into a Arduino Nano :+1:   https://github.com/tvixen/diy-BMS-with-Arduino-Nano

Br Tim

chickey, 2018-07-17:
Have you burnt the fuses on the attiny85?

Rorschac, 2018-07-17:
Yes. I designed my own board with altium based on orginal schematic and component for product PCB at home. I want to share this design if I can run it. I bootloaded orginal code in fourteen attiny85. Just I changed ESP code for load uno or mega. I can send message with uno on I2C but no response attiny85 board.

chickey, 2018-07-17:
Ok asking as I had no luck initially and had to increase the stretch clock from 1000 to 2500 as memory serves as I couldn’t get any communication either but I hadn’t set the fuses so the speed etc was incorrect and the stretchclock increase kind of masked the issue.

tvixen, 2018-07-17:
You can use Stuart's slave software, and then try mine for your master (UNO) This should work.

Rorschac, 2018-07-17:
I used your code but no difference. Uno sends but attiny85 not talk. I see the signal at the output of ADUM but attiny is not talk.

Merdeka, 2018-07-17:
Haven't tested it but could it be an issue with the default i2c bus speed? 100khz vs 400kHz.

Cheers,
Timo

Rorschac, 2018-07-17:
I found a ESP8266 and used the orginal code, original bus speed. It doesn't work too. Attiny85 doesn't reponse again. I controlled everything on pcb. It seems normal. I don't know why not running it?

tvixen, 2018-07-17:
Do you see the green diode blink on the Cell_module. If not, check your programming of the ATTiny85.

tvixen, 2018-07-17:
If you use Uno, you don't have to use Optiboot. As it's only Nano having a problem with the watchdog.

tvixen, 2018-07-17:
Try ONLY to connect one module to the bus. When this is working, add another. aso aso.

chickey, 2018-07-17:
Also try the clock stretch change

Rorschac, 2018-07-17:
Led is not blinking. But I saw a video on youtube, diyBMS tutorial it haven't blinking too. Already I'm working with only one cell.  Also I controlled reset pin voltage, it's almost 2V.

https://www.youtube.com/watch?v=ZahB-86zzfQ

chickey, 2018-07-17:
https://youtu.be/CbuOteAcv04

Rorschac, 2018-07-17:
Thanks, I have watched video now.  I guess, your github code have all changes from your video. I will try now.

Rorschac, 2018-07-17:
Did you added 2.2K pull-up resistors to esp I2C wire, except modul?

tvixen, 2018-07-17:
Yes this is very important to do. If not, there's no 'power' to feed the bus.

Rorschac, 2018-07-18:
I loaded Colin's codes. SCK does not seem normal, does it? It's a bit short.

Scope1-SCK
Scope2-SDA

[image]

chickey, 2018-07-18:
Module code should be identical and I backed out the stretch clock changes in the controller code.  I more put the video to show the issues I had as I hadn’t done the fuses etc

Rorschac, 2018-07-18:
Is it possible to communicate with the scope as it looks? As a result, everybody have worked. Should I search the fault somewhere else? I know I'm starting to get boring, I'm sorry for that but I need help.

Rorschac, 2018-07-18:
There is a warning in the code. Could the factory have set for external crystal?
Because the module does not work even in the simplest code like led blink.

`If you burn incorrect fuses to ATTINY85 you may need to connect a crystal over the pins to make it work again!`

Rorschac, 2018-07-18:
I found something like this. I hope this is the problem.
[](http://www.instructables.com/id/Simple-and-cheap-Fuse-Doctor-for-Attiny)

Rorschac, 2018-07-19:
I have a good and bad news for you. The good news is that the problem is solved, the bad news is that I think the bootloading code is loading. I hit the upload button and the problem is solved. And Colin, your arrangements are wonderful. I thank you all very much.

closed 2019-03-04 by stuartpittaway