stuartpittaway/diyBMS #14: BOM updated

ArchGalileu, 2018-09-19:
Hello

Which BOM? from README or from wiki?

Does it make difference from Colin Hickey?

Any chances to have a list from farnell? 

I'm new to this, more or less, so to not mistake better to ask ;)

Best regards
ArchGalileu

stuartpittaway, 2018-09-19:
Hi ArchGalileu, the circuit in Colin's branch is the same as this one - its only the code thats different.

I've not got a farnell list I'm afraid - but most of the components are available from most suppliers.  I got mine all through AliExpress for cost reasons.

KrX3D, 2018-09-26:
hello,
i bought mine from readme, where is the wiki and whats the difference?

ArchGalileu, 2018-09-26:
My mistake... from README is correct ;)

fjgaston, 2019-02-20:
Hi Stuart,
I'm ordering parts at the moment, as I'm going to build your BMS for my battery.
I noticed the Drain Resistor is noted for 50 Watts it seems to be huge, did you mean 5 Watts.
From my calculations with 4.2 Volts and a 3R3 resistor the power would be 5.35 Watts so I guess a 7W should be enough ?
For the ADUM1250, do you think the MAX14933 could be a good less expensive substitute ?

Zbysekz, 2019-02-20:
7 Watts will go very very hot... i bought these on aliexpress. https://imguh.com/image/oIHBh

Zbysekz, 2019-02-20:
And i have used ISO1540DR, it is more available in EU and the price is also good. I will post complete schematics etc.. when it will be finished

fjgaston, 2019-02-20:
I think I will go for a 10 Ohms resistor instead, the maximum will be 1.764 Watts so I guess a 10 Watts wirewound resistor like this one [https://fr.farnell.com/multicomp/mcprw0awjw100b00/wirewound-resistor-10-ohm-10w/dp/1737549](https://fr.farnell.com/multicomp/mcprw0awjw100b00/wirewound-resistor-10-ohm-10w/dp/1737549) should be more than enough.
The BOM in the readme says components cost is around £5.50 but on farnell a 3R3 50 Watts resistor is more than 6£ :-/
Anyway the 10 ohms resistor will be enough for the moment for small balancing.
I ordered ADUM1250s on Aliexpress (x5 times cheaper this is crazy) but I'll also order a MAX14933 to test the difference (it's 1.96€ on farnell vvs 5.47€ for the ADUM1250 and 4.53 for the ISO1540) there is also the SI8600AB taht is 2.94€ I wonder if they would be all be compatible?

stuartpittaway, 2019-02-20:
> 7 Watts will go very very hot... 

Agreed it will get hot!

stuartpittaway, 2019-02-20:
> The BOM in the readme says components cost is around £5.50 but on farnell a 3R3 50 Watts resistor is more than 6£ :-/

Yes although I brought a lot of items from AliExpress - take a look at...

https://www.aliexpress.com/item/Novel-Auto-Car-Vehicle-4Pcs-Professional-50W-Load-Resistor-6ohm-Fix-LED-Bulb-Fast-Hyper-Flash/32907694269.html?spm=2114.search0104.3.2.511d5acbVymNHO&ws_ab_test=searchweb0_0,searchweb201602_8_10065_10068_10130_10547_319_317_10548_10696_10192_10190_453_10084_454_10083_10618_10307_10820_10301_10821_10303_537_536_10902_10059_10884_10887_321_322_10103,searchweb201603_57,ppcSwitch_0&algo_expid=00d20ee5-5890-4a57-ad6a-245f0c2d67bd-0&algo_pvid=00d20ee5-5890-4a57-ad6a-245f0c2d67bd&transAbTest=ae803_5

or search for "Novel Auto Car Vehicle 4Pcs Professional 50W Load Resistor 6ohm Fix LED Bulb Fast Hyper Flash Turn Signal Light Bulb"

$3 for 4x 50W resistors

fjgaston, 2019-02-20:
thank you for the infos, I just forked this project and building one now, I'll keep you updated :-)

Church1182, 2019-02-20:
I just ordered parts for mine and mixed Mouser and AliExpress parts with boards from www.allpcb.com. I'm sitting at a cost per board for 15 boards of $8 US. I'm sure that can be lower if you take the time to source all parts through AliExpress and eBay sites.

fjgaston, 2019-02-20:
sourcing parts takes so much time (and it's much less interesting than building or programming)