stuartpittaway/diyBMS #18: why 3.3V for the ATTiny ?

fjgaston, 2019-02-24:
Hi, sorry if this question is not relevant, I'm just a beginner, why do you use a 3.3V buck/boost converter and not a 5V boost converter to power the ATTiny ?

stuartpittaway, 2019-02-24:
Hello, you could use a 5v boost converter however as we communicate with the esp8266 that needs 3.3v via the adum chip.

fjgaston, 2019-02-24:
I though the logic voltage on each side of the ADUM1250 was relative to the VDD, I though it could have 5V on one side and 3.3V on the other side

stuartpittaway, 2019-02-24:
Yes you are correct, I had dumb moment there!!

fjgaston, 2019-02-24:
Yet I don't think it is dumb cause while you were in the prototyping process, I think you had to test the module before putting the isolator so it made more sense to have the same voltage as the ESP.

closed 2019-03-26 by stuartpittaway