stuartpittaway/diyBMS #20: J1, C6 and C1 are very close on the PCB (v2.1)

fjgaston, 2019-03-01:
Hi,
I just started oldering my first PCB and I noticed that C6 and C1 are bigger than the footprint (I ordered a 1210 smt version as mentioned in the readme) so it's not easy to solder J1, C6 and C1 without having them touching each others.
Adam Welch also mentioned it in his video [https://youtu.be/ETw4SNH7SXI](https://youtu.be/ETw4SNH7SXI) 

 From I see there sould be enough room on the PCB to put C6 and C1 lower.

stuartpittaway, 2019-03-04:
Hello, I can see that C6 and C1 are close together - I use solder paste so the capacitor sits on top of the pads.

You mention J1 - this should be easy to solder as its on the other side of the board?

fjgaston, 2019-03-04:
I mentioned J1 but in fact I wanted to mention the REG710-3.3
Maybe I should have ordered smaller capacitors.
Anyway I'll try to draw a gerber file and if I succeed, I'll send it to you if you want.

stuartpittaway, 2019-03-04:
Did you see the smaller PCB already in use? Its in the "small" branch.  This has probably already fixed the issues you have.

Zenkormalorvi, 2019-04-01:
Hi, I am also experiencing some issues with the capacitors being really close together and the 1210 size units extending to the extreme ends of both pads basically making it impossible to solder with an iron.  I ordered my boards on mar 7 2019 so it should have been updated from the previous user comments.  I appreciate everything you guys have done with this project. Thanks so much

fjgaston, 2019-04-01:
Did you use the smaller PCB or the standard?
I had also troubles with the first two boards I soldered, pin 6 of the REG710 was badly soldered and I had to de-solder C6 to re-solder it. It's funny that the same thing happened to me twice on the same pin of the same component. Maybe the pads could be a bit wider on the PCB? Maybe it works better with the soldering paste method but I don't have soldering paste.

aussiegwapo, 2019-04-01:
Did you use solderpast or solder the old fashion way with wire and iron.
Could even try hot fry pan to settle the solder evenly.

stuartpittaway, 2019-04-01:
Okay, I can see the confusion.  People are building the smaller PCB using the component list from the larger version.

All parts on the smaller pcb are 0805 sized, not 1210 like the older larger board.

I recently removed the larger board from github so hopefully this problem goes away for new starters.

fjgaston, 2019-04-01:
Soldered the old fashion way this the soldering iron, I don't feel comfortable with the solder paste.
Thank you fir the feedback Stuart and thank you for this incredible project.

stuartpittaway, 2019-04-09:
Yes you can hand solder 0805 - you will need tweezers or similar to hold the component.

Some times its easier to put a small blob of solder on each pad and then press the component onto the top and touch each pin/leg with the iron to remelt the solder pad.

Zenkormalorvi, 2019-04-09:
Yes you can solder all the components on the smaller pcb design including the 0805 size ones with a soldering iron. I just finished soldering mine a few days ago and it is working. I recommend putting solder on 1 pad only and then using tweezers to position the component. once the component is positioned correctly and soldered on 1 pin then continue soldering the other pins. also, flux is your friend.