stuartpittaway/diyBMS #22: ESP32 diyBMS

fjgaston, 2019-03-31:
Hi,
I would like to use an ESP32 instead of the ESP8266, did anyone tried it ? do you have some advice to help me to port the code for an ESP32 ?

stuartpittaway, 2019-04-02:
Hello, I would recommend sticking with the 8266 - it works fine, you dont need an ESP32 for this to work and once you start changing the code you wont get the support or fixes other may contribute

fjgaston, 2019-04-03:
I don't have an ESP8266 at the moment and I though they were very similar and the code would be very easy to port but I just got into the code and that doesn't seem to be that easy. I'll start with a heavily simplified version (with no gui) at the moment until I receive the ESP8266.

gpmontt, 2019-06-04:
Actually, the integration with the module esp32 is a bad idea, because this, have a problem with the [precision](https://github.com/espressif/esp-idf/issues/164#issuecomment-299669907).

[image]