stuartpittaway/diyBMS #25: PWM driven mosfet onstead of resistor (not an Issue)

uploader18, 2019-04-07:
Hi!

Since in my packs built from used 18650 there is no use for small balancing currents as they are not all same capacity, I was wondering if you were thinking about using mosfet as dummy load or even a flyback converter for regenerative balancing?

Regards Žiga

stuartpittaway, 2019-04-07:
Hi, I've no plans to move away from resistor based load balancing - it works for 18650 cells, what do you use?

uploader18, 2019-04-07:
For now I use flyback converter that boosts 4.2V to 48V. I am charging my packs with 80Amp and I need at least 10Amps of balancing current. My system is very simple...when cell reaches 4.1V it starts balancing with input current limit to primary side and no voltage regulation on secondary side(48V).

fjgaston, 2019-04-07:
I've seen somewhere a project where a DC/DC converter is used for balancing, I don't remember the website or the name.
Having the same capacity on every cell is important because your battery's capacity will be your smallest cell's capacity.
If your cells don't have the same capacity you can use more cells in parallel to better match the capacity.

uploader18, 2019-04-07:
I have packs 100p 14s but since there are mix cells with mixed capacity and degradation it is almost impossible to get them all equal, you can get close but once you have 200Ah packs and there is 20Ah difference you have a problem with unbalance. So it is easier for me ta make an active balancer. I would spend a lot of time measuring capacity of every pack and then add cells.

fjgaston, 2019-04-07:
just take some cells from the biggest pack and put them in the smallest pack

uploader18, 2019-04-07:
At fist i did just that, but i spot welded fuses on batteries and disassembling whole pack since i have 5x4 holders is a nightmare.

uploader18, 2019-04-07:
This guy made a nice balancer.
https://hackaday.io/project/20825-low-cost-non-dissipative-active-battery-balancer