stuartpittaway/diyBMS #27: Confused

mikegleasonjr, 2019-04-13:
Hi, going through the README it is not obvious of the master branch is the smaller PCB variant. Since the smallpcb branch is several commits behind master, which is which?

Also, I heard that a contributor (Colin?) made several updates, are they incorporated here or his fork more up to date?

In short, there is no clear BOM and instructions. Or do I miss something?

stuartpittaway, 2019-04-13:
Hello, the current master branch is the correct one.  Colin did make some code changes, and they were merged in to the master branch recently.

mikegleasonjr, 2019-04-14:
Thanks will try to gather everything up to assemble your BMS and create a PR if needed for the docs!

closed 2019-04-14 by mikegleasonjr