stuartpittaway/diyBMS #32: Access to the interface when no internet connexion

fjgaston, 2019-07-29:
Hi,
I can't connect to the interface when I have no internet.
I see it tries to access https://stuartpittaway.github.io/diyBMS/loader.js
It can be a problem mostly when having a mobile system (or being somewhere with unreliable internet)
Is there a way to easily solve this ?

stuartpittaway, 2019-07-29:
Hello, unfortunately not at this time.  It was designed this way so that the BMS is always using the latest web interface, however the downside is that an Internet connection is required.

The Version 4 DIYBMS doesn't have this issue as all the code is on the WeMOS.

A similar code change could be made on the V3, but I won't be doing this myself but happy to accept a pull request from others.

closed 2019-07-29 by stuartpittaway

fjgaston, 2019-07-29:
thank you for the reply, I'll go for the v4 for my next build, it seems to have many other advantages.

acidice333, 2021-01-24:
i have made a fork with code updates to work offline.
https://github.com/acidice333/diyBMS

you will have to upload the file system once you've uploaded the new Controller code.
see https://github.com/earlephilhower/arduino-esp8266littlefs-plugin for details on doing that.

stuartpittaway, 2021-01-24:
Thanks for this, this is for version 3 of the diybms, which has been obsolete for a long time now, but thanks for taking the time to add new features.